machinekit/machinekit-docs #12: I'd _love_ to be able to view beautiful `.hal` source highlighted in the `.asciidocs`

luminize, 2015-01-15:
When you add info about source code highlighting in `.asciidoc` files like example below it will highlight the code on the github page. From reading point of view I would love to be able to view `.hal` files in the same way as we can view `.ini` files
```
[source,ini]
----
[PRUCONF]
DRIVER=hal_pru_generic
CONFIG=pru=1 num_stepgens=4 num_pwmgens=3
PRUBIN=xenomai/pru_generic.bin
----
```
in (this issue example) markdown it would be done in this way:
```
```ini
```
which gives:
```ini
[PRUCONF]
DRIVER=hal_pru_generic
CONFIG=pru=1 num_stepgens=4 num_pwmgens=3
PRUBIN=xenomai/pru_generic.bin
```

luminize, 2016-05-27:
@mhaberler did I remember correctly that there is a possibility for syntax highlighter for hal files in the new docs?

mhaberler, 2016-05-30:
yes - it needs a pygments extension for HAL, protobuf and G-code

I have started on it, but it is unfinished

back then I did the same to source-highlight, that is still in the repo here: https://github.com/machinekit/machinekit-docs/tree/master/docs/source-highlight

luminize, 2016-06-20:
#171 just shows hal source file highlighting! Very cool!

closed 2016-06-20 by luminize