machinekit/machinekit-docs #21: inadequate information on how people should report an error on the list

luminize, 2015-06-15:
instead of an obscure question from a user developers frequently need more info, like a pastebin of linuxcnc.log, info about the system, platform etcetera.

We need to document this and point to this when the info provided is inadequate. So then without that info the user has to follow a quick tut of list on what to provide in what form before we can continue.

mhaberler, 2015-06-15:
we definitely need a landing page for reporting an error, the repititiveness of interactions is nauseating

I did this a while ago but its completely outdated; https://github.com/mhaberler/asciidoc-sandbox/wiki/Reporting-an-error-with-Machinekit - maybe a starting point

ideally we also would have a platform/config/log reporting script which uploads the whole story onto a temporary website - there are some open source examples but nothing really appealing

luminize, 2015-07-08:
@mhaberler could you add the [mentioned link](https://github.com/mhaberler/asciidoc-sandbox/wiki/Reporting-an-error-with-Machinekit) to your local machinekit-docs repo, just put it somewhere. I'll pull that branch, clean it up and put it in machinekit-docs. A script is a nice to have, but until then let's not procrastinate and put the effort of getting a proper report at the user.

luminize, 2015-08-03:
closed via #25 and #26

closed 2015-08-03 by luminize