machinekit/machinekit-docs #23: wiki : no picture

LETARTARE, 2015-07-07:
Hello,
I have no picture on my browser  'Seamonkey 2.33.1' ?
Kind regards

luminize, 2015-07-07:
@LETARTARE i fail to see what you mean with this issue.

LETARTARE, 2015-07-07:
Oh, yes, here is a copy of the result:
[image]
Figure 2 does not appear !

luminize, 2015-07-07:
ok, don't use the wiki.
you need to view the repo itself, like here:
https://github.com/machinekit/machinekit-docs/blob/master/src/common/user_intro.asciidoc

if this solves your issue, would you close it?
thanks

LETARTARE, 2015-07-07:
Do I try to change the path of Wiki images ?

closed 2015-07-07 by LETARTARE

zultron, 2015-07-07:
@luminize, did you just shut off the wiki?  I can't even bring up the OP's link.  Good!  (But I'm confused)

luminize, 2015-07-07:
@zultron yes i did (disable wiki). Imo it has no use since the wiki is yet another repo. And contributions should be made in the repo itself and not the wiki.

zultron, 2015-07-07:
Great!  I did this on other MK repos at one point, but maybe the MK-docs wiki was created later.  I'll make another check.  Thanks, Bas.

luminize, 2015-07-07:
Hi John,
Yes it was. I transferred this repo from my account to the MK organization. I dabbled with the Wiki in the beginning when I got the docs out of the MK code base and I guess this was a leftover from that experiment.
One of the problems with the wiki was as the OP mentioned that the pictures couldn't be displayed.
Bas