machinekit/machinekit-docs #38: Installation on Jessie instructions are missing

luminize, 2015-09-26:
When all is mature enough for Machinekit to be supported on Debian Jessie, the installation instructions should be updated.

reference machinekit  issue : https://github.com/machinekit/machinekit/issues/598

luminize, 2015-09-26:
@ArcEye @mhaberler 
These comments are for reference so we can continue in this issue as to keep the focus re. documentation here
1. https://github.com/machinekit/machinekit/issues/598#issuecomment-143419734
2. https://github.com/machinekit/machinekit/issues/598#issuecomment-143445896
3. https://github.com/machinekit/machinekit/issues/598#issuecomment-143466492

What's not done from nr. 1 is that I don't know how to make a preference of the dovetail automata repo over `sid`.

regarding nr. 3 I guess that when you are building kernels you have more knowledge, and we can put that in the instructions when you build a development platform.
also regarding nr. 3, do we need to split out the kernels installation instruction if they are not identical?

ArcEye, 2015-09-27:
>What's not done from nr. 1 is that I don't know how to make a preference of the dovetail automata repo over sid

As this Debian guide alludes to, there can be complex problems with preferences, especially if you have to use more than 2 repos
`https://wiki.debian.org/AptPreferences`

I have used them but once got into a lot of problems because of dependencies that a package from a low preference repo had that were pulled from a higher preference repo.  
That is why I stopped using it and just have commented out extra repos, which I uncomment and do specific pulls using the `-t repo-name` in the apt-get stanza.

luminize, 2015-09-27:
I did some searching and came to this: http://unix.stackexchange.com/questions/107689/how-to-install-a-single-jessie-package-on-wheezy where a single package is pinned.

Reading this http://linux.die.net/man/5/apt_preferences I understand that the order in which repositories are mentioned in the `sources.list` determine priority _if_ there is no `preferences` file. It's also possible to pin a specific version...

All in all, commenting out the sid sources after installing libczmq an libczmq-dev from `sid` is a small thing.

ArcEye, 2015-09-27:
>regarding nr. 3 I guess that when you are building kernels you have more knowledge, and we can put that in the instructions when you build a development platform.
also regarding nr. 3, do we need to split out the kernels installation instruction if they are not identical?

See PR #39

ArcEye, 2015-10-07:
Re #598 (issuecomment-145959066) so it is on this thread too

I would suggest however creating a file within /etc/apt/preferences.d
It is a cleaner solution and the reason for the additions is obvious from the file name.

I used your file contents in a file called /etc/apt/preferences.d/machinekit-jessie
After running update, I verified by deleting firmware-realtek, which I know has a more up to date version in sid.
apt-get install firmware-realtek installs the jessie non free archive version
apt-get install -t sid firmware-realtek installs the updated sid version

luminize, 2015-10-08:
Thx, I'll rework the preference file from your remark. Thanks for reviewing.

PR #49 (2016-01-06): Jessie instructions 2
This PR should contain all info on
- setting up on debian Jessie
- building an RT-PREEMPT 

needed for https://github.com/machinekit/machinekit/issues/598
fixes #38
closed 2016-01-06 by jpmoniz