machinekit/machinekit-docs #42: libzmq3-dev installation fails on wheezy

sirop, 2015-11-04:
Referring to https://github.com/machinekit/machinekit-docs/blob/master/machinekit-documentation/developing/machinekit-developing.asciidoc
```
$ sudo apt-get install libzmq3-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following package was automatically installed and is no longer required:
  libzmq1
Use 'apt-get autoremove' to remove it.
The following extra packages will be installed:
  libzmq3
The following NEW packages will be installed:
  libzmq3 libzmq3-dev
0 upgraded, 2 newly installed, 0 to remove and 17 not upgraded.
Need to get 0 B/876 kB of archives.
After this operation, 1,914 kB of additional disk space will be used.
Do you want to continue [Y/n]? Y
(Reading database ... 184756 files and directories currently installed.)
Unpacking libzmq3:amd64 (from .../libzmq3_3.2.3+dfsg-2~bpo70+1_amd64.deb) ...
dpkg: error processing /var/cache/apt/archives/libzmq3_3.2.3+dfsg-2~bpo70+1_amd64.deb (--unpack):
 trying to overwrite '/usr/lib/x86_64-linux-gnu/libzmq.so.3', which is also in package libzmq4:amd64 4.0.4-2~1da~wheezy1
Selecting previously unselected package libzmq3-dev:amd64.
Unpacking libzmq3-dev:amd64 (from .../libzmq3-dev_3.2.3+dfsg-2~bpo70+1_amd64.deb) ...
dpkg: error processing /var/cache/apt/archives/libzmq3-dev_3.2.3+dfsg-2~bpo70+1_amd64.deb (--unpack):
 trying to overwrite '/usr/lib/x86_64-linux-gnu/libzmq.a', which is also in package libzmq4-dev:amd64 4.0.4-2~1da~wheezy1
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/libzmq3_3.2.3+dfsg-2~bpo70+1_amd64.deb
 /var/cache/apt/archives/libzmq3-dev_3.2.3+dfsg-2~bpo70+1_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

but:
```
$ sudo apt-get install libzmq4-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libpgm-5.1-0 libzmq1
Use 'apt-get autoremove' to remove them.
The following NEW packages will be installed:
  libzmq4-dev
0 upgraded, 1 newly installed, 0 to remove and 17 not upgraded.
Need to get 878 kB of archives.
After this operation, 2,164 kB of additional disk space will be used.
Get:1 http://deb.dovetail-automata.com/ wheezy/main libzmq4-dev amd64 4.0.4-2~1da~wheezy1 [878 kB]
Fetched 878 kB in 0s (1,299 kB/s)
Selecting previously unselected package libzmq4-dev:amd64.
(Reading database ... 184695 files and directories currently installed.)
Unpacking libzmq4-dev:amd64 (from .../libzmq4-dev_4.0.4-2~1da~wheezy1_amd64.deb) ...
Processing triggers for man-db ...
Setting up libzmq4-dev:amd64 (4.0.4-2~1da~wheezy1) ...
```
works.

strahlex, 2015-11-04:
As the error message says: remove libzmq4-dev and all its dependencies then install libzmq3-dev. There is a naming issue in wheezy meaning that the ZMQ libraries are named incorrectly.

strahlex, 2015-11-04:
But I agree that the document is confusing since it mentions wheezy and jessie.

sirop, 2015-11-04:
But which is better libzmq4-dev or libzmq3-dev ?

strahlex, 2015-11-04:
As I can recall the naming issue started with Jessie. So libzmq3-dev on Jessie is equal to libzmq4-dev for Wheezy from zultrons repositories. If you have wheezy go for libzmq4-dev.

See also:
http://www.machinekit.io/docs/building-from-source/

sirop, 2015-11-04:
Indeed: just built successfully from source with **libzmq4-dev**  installed on wheezy.

closed 2015-11-04 by sirop