machinekit/machinekit-docs #43: document beaglebone dev setup in docs from linuxcnc wiki

luminize, 2015-11-10:
Just saw this link appear on the list.
http://wiki.linuxcnc.org/cgi-bin/wiki.pl?BeagleboneDevsetup

In due time we should pick these instructions and put them into the docs at the appropriate place.

RunningLight, 2015-11-11:
Setting aside the fact that this page is nearly 3 years out of date, it is really about creating a convenient environment for software development on any limited-resource board (which I'll call an "sbc" below) and not just on the BeagleBone.

Among the principal topics are

* setting up a remote DHCP/TFTP server
* setting up a remote NFS server
* configuring Das Boot on the sbc so it will boot and/or run from the remote servers instead of its local storage (eMMC, uSD, whatever) 
 * this is the only topic which seems likely to depend on the specific sbc make and model
* running headless
* adding swap space

I contributed many of the annotations to Michael's original LinucCNC Wiki page and I'm willing to create a first cut of these topics---probably as a set of docs---for the machinekit-docs repo. A related topic is the more recent work by others to distribute compilations, which I'll try to weave in with this work as well so we have a coherent story to tell interested developers.

luminize, 2018-10-09:
3 years ago, Images are made by RCN, no real need for this info since this is a Beaglebone thing (we don't show how to set up an RPI or a PC for that matter. Closing.

closed 2018-10-09 by luminize