machinekit/machinekit-docs #50: machinekit-xenomai: /usr/lib/rtapi.so  missing?

sirop, 2016-01-05:
Referrering to http://www.machinekit.io/docs/packages-debian/ .

Just tried to install machinekit-xenomai on an identical machine as I did it two months ago:
```
~$ uname -a
Linux debian-slave 3.8-1-xenomai.x86-amd64 #1 SMP Debian 3.8.13-12~1wheezy~1da x86_64 GNU/Linux
~$ halrun
msgd:0 stopped
rtapi:0 stopped
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded
halcmd: hal_init() failed: -22
NOTE: 'rtapi' module must be loaded
```
So `rtapi.so` cannot be found somehow...  

```
~$ sudo find / -iname rtapi.so
/usr/lib/pyshared/python2.7/machinekit/rtapi.so
/usr/lib/linuxcnc/posix/rtapi.so
/usr/lib/linuxcnc/xenomai/rtapi.so
/usr/lib/python2.7/dist-packages/machinekit/rtapi.so
```

But the same on the other machine (same hardware) yields:
```
~$  sudo find / -iname rtapi.so
/usr/lib/rtapi.so
/usr/lib/linuxcnc/xenomai/rtapi.so
/usr/lib/linuxcnc/posix/rtapi.so
/usr/lib/pyshared/python2.7/machinekit/rtapi.so
/usr/lib/python2.7/dist-packages/machinekit/rtapi.so
```  
So the older installtion has `/usr/lib/rtapi.so` .
This older machinekit-xenomai installation is of 1st Nov 2015.

ArcEye, 2016-01-06:
Hi,
This issue tracker relates to the github documentation, not the website instruction page.
It will get missed here.

Could you cut and paste your entry here
`website out of date post build/CI infrastructure change #840`
in the main machinekit repo issue tracker, and close this issue

The packages are being looked at in that issue and your problem can be added to them

Thanks

sirop, 2016-01-06:
Thanks for the hint.  

Closed.

closed 2016-01-06 by sirop