machinekit/machinekit-docs #60: Sid package on Jessie?

strahlex, 2016-03-07:
Why do we need this `libczmq` package on Jessie? https://github.com/machinekit/machinekit-docs/blob/master/machinekit-documentation/getting-started/APT-packages-jessie.asciidoc#add-the-sid-repository-to-you-debian-sources-list-and-update

luminize, 2016-03-07:
@strahlex please see this comment (and if you're up to it the history leading to that :) ) https://github.com/machinekit/machinekit/issues/598#issuecomment-143419734

sirop, 2016-05-02:
@luminize 
Then this line shouls not contain any `libczmq` as this `libczmq` would come from jessy:
``` 
sudo apt-get install libczmq-dev python-zmq libjansson-dev \
  libwebsockets-dev libxenomai-dev python-pyftpdlib cython bwidget lsb-release
```
https://github.com/machinekit/machinekit-docs/blob/master/machinekit-documentation/developing/machinekit-developing.asciidoc#install-packages-required-for-building-from-source

luminize, 2016-05-02:
@sirop @strahlex
It was needed at the time. I wrote down gathered instructions from https://github.com/machinekit/machinekit/issues/598 with @mhaberler and @ArcEye for those instructions were needed as part of that thread if i recall correctly.

If there are better instructions (you might know better than me) it can be updated.

mhaberler, 2016-05-02:
I think @sirop is right for jessie

sirop, 2016-05-03:
@luminize @mhaberler 
If one follows this advice
>The easiest way to fetch all the machinekit runtime packages is to install a current package, and then delete it     

`libczmq2:amd64   2.2.0-0.6~1jessie~1da` gets anyway installed, but does not seem to disturb.
```
~$ dpkg -l | grep libczmq
ii  libczmq-dev:amd64                      3.0.2-7                                           amd64        High-level C binding for ZeroMQ (development files)
ii  libczmq2:amd64                         2.2.0-0.6~1jessie~1da                             amd64        High-level C binding for ZeroMQ
ii  libczmq3:amd64                         3.0.2-7                                           amd64        High-level C binding for ZeroMQ
```
`make` runs without errors and halrun can be started.  

Is there some ultimative test for `libczmq` within machinekit?

mhaberler, 2016-05-03:
@sirop "ultimate test" like "being used"? yes, everytime you say 'halcmd loadrt ...' etc; Or haltalk.

right, the 2.2 version is superfluous; luckily 2.2 and 3.0 are API compatible

strahlex, 2016-06-20:
Has been removed.

closed 2016-06-20 by strahlex