machinekit/machinekit-docs #71: Missing supporting packages on jessy

sirop, 2016-05-03:
I followed this advice:
> The easiest way to fetch all the machinekit runtime packages is to install a current package, and then delete it  

but  `./configure`  told me that some packages are still missing beginning with _Protobuf compiler_.

So I had to install missing packages one by one until `./configure` did not complain any  more.  
Those were:
```
  sudo apt-get install -y  libprotobuf-dev protobuf-compiler python-protobuf libprotoc-dev
  sudo apt-get install -y  libtool liburiparser-dev
  sudo apt-get install -y  libssl-dev  openssl
  sudo apt-get install -y  uuid-dev
  sudo apt-get install -y  libavahi-client-dev
  sudo apt-get install -y  libmodbus-dev
  sudo apt-get install -y  libusb-1.0-0-dev
  sudo apt-get install -y  libreadline-gplv2-dev
  sudo apt-get install -y  libglib2.0-dev
  sudo apt-get install -y  libgtk2.0-dev tcl8.6-dev tk8.6-dev
  sudo apt-get install -y  libxaw7-dev
  sudo apt-get install -y  libboost-python-dev
  sudo apt-get install -y  libglu1-mesa-dev
```  
The order of installtion _may be_ important.

I did then
```
sudo apt-get remove --purge machinekit
```
after `make` run.

mhaberler, 2016-05-03:
this advice is about **runtime packages**, not **required-for-build** packages

If you follow this: https://github.com/machinekit/machinekit-docs/blob/master/machinekit-documentation/developing/machinekit-developing.asciidoc

and do the `sudo mk-build-deps -ir` step, it'd all be covered. Why the hoops?

sirop, 2016-05-03:
@mhaberler 
I followed https://github.com/machinekit/machinekit-docs/blob/master/machinekit-documentation/developing/machinekit-developing.asciidoc  
with the only difference that I built only for `posix` and `xenomai`.
So I did:
```
debian/configure -px
sudo mk-build-deps -ir
```
Is it the absent `r` for _rt-preemt_ in `debian/configure -px` that makes me miss me some packages?

mhaberler, 2016-05-03:
please describe exactly what base system you have and all the steps you took, including error messages - "makes me miss some packages" is not an error report somebody can act upon

sirop, 2016-05-03:
I began with a fresh Debian install based on `debian-8.3.0-amd64-CD-1.iso`.
No desktop  - gnome or similar - was installed. Openssh server was installed.  

Then:
```
sudo apt-get install linux-image-xenomai.x86-amd64
sudo apt-get install linux-headers-xenomai.x86-amd64
sudo reboot
```
From now on under `3.8-1-xenomai.x86-amd64 jessy`:
```
sudo sh -c \
"echo 'Package: *
Pin: release a=stable
Pin-Priority: 900

Package: *
Pin: release o=Debian
Pin-Priority: -10' > \
/etc/apt/preferences.d/sid;"
```
Edited `/etc/apt/sources.list` .  
```
sudo apt-get update
sudo apt-get install -t sid libczmq-dev
sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv 43DDF224
sudo sh -c \
  "echo 'deb http://deb.machinekit.io/debian jessie main' > \
  /etc/apt/sources.list.d/machinekit.list"
sudo apt-get update
```

As I thought that I'd need _to fetch all the machinekit runtime packages_ ,
I did at this point: 
```
sudo apt-get install machinekit
```

Then:
```
sudo apt-get install git dpkg-dev
sudo apt-get install --no-install-recommends devscripts equivs
git clone https://github.com/machinekit/machinekit.git
cd machinekit

debian/configure -px
sudo mk-build-deps -ir
cd src
./autogen.sh
./configure --without-rt-preempt  --with-posix  --with-xenomai
```
Now `./configure` complained about missing packages which I had to install one by one 
as above said in https://github.com/machinekit/machinekit-docs/issues/71#issue-152714271 .

mhaberler, 2016-05-03:
can you reproduce this including a log?

there's definitely something wrong

example: liburiparser-dev  - stock from debian stream, and a build dep here: https://github.com/machinekit/machinekit/blob/master/debian/control.in#L15

`sudo mk-build-deps -ir`MUST catch this one

sirop, 2016-05-03:
I have to setup a fresh jessy installation for that. Will do it tomorrow morning.

strahlex, 2016-05-03:
@sirop I can recommend using Vagrant for testing such things. (see https://github.com/strahlex/machinekit-vagrant)

sirop, 2016-05-04:
@strahlex 
I do not know if you are the right person to complain about it,
but xenomai on jessie 8.4.0 cannot be started with Vagrant/VirtualBox.
There is a kernel panic at the beginning of the start.
My ` /etc/default/grub` is
```
GRUB_DEFAULT="1>2"
GRUB_TIMEOUT=0
GRUB_DISTRIBUTOR=Debian
GRUB_CMDLINE_LINUX_DEFAULT="quiet xeno_nucleus.xenomai_gid=120"
GRUB_CMDLINE_LINUX=""
```

Anyway I'll do the same tomorrow on a "real" machine.

sirop, 2016-05-05:
@mhaberler 
Ok, I just set up  _debian-8.3.0-amd64-CD-1.iso_ on a "real" machine.
I began with these packages installed as `dpkg-query -l ` shows them:
https://gist.github.com/sirop/7984794ba4946e3c14d81be94bbc0f8b

Then I did:
```
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv 43DDF224
sudo sh -c \
  "echo 'deb http://deb.machinekit.io/debian jessie main' > \
    /etc/apt/sources.list.d/machinekit.list"
sudo apt update
sudo apt -y install linux-image-xenomai.x86-amd64 linux-headers-xenomai.x86-amd64
```
Recorded the terminal session with `script`:
https://gist.github.com/sirop/f31a7e818d71445c7a4e540873d8f3b0

The only difference is my `debian/configure -px` instead of `debian/configure -prx`.
As you see, `./configure` yields:
```
checking for CZMQ... yes
checking for PROTOBUF... no
configure: error: Package requirements (protobuf > 2.4.0) were not met:

No package 'protobuf' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables PROTOBUF_CFLAGS
and PROTOBUF_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
```  

`dpkg-query -l | grep protobuf` does not find anything.

ArcEye, 2016-05-06:
The protobuf packages I have installed on jessie rt-preempt amd64 are:
```
libprotobuf-c-dev_1.0.2-1_amd64.deb
libprotobuf-c1_1.0.2-1_amd64.deb
libprotobuf-dev_2.6.1-1_amd64.deb
libprotobuf-lite9_2.6.1-1_amd64.deb
libprotobuf9_2.6.1-1_amd64.deb
libprotoc-dev_2.6.1-1_amd64.deb
libprotoc9_2.6.1-1_amd64.deb
protobuf-c-compiler_1.0.2-1_amd64.deb
protobuf-compiler_2.6.1-1_amd64.deb
python-protobuf_2.6.1-1_amd64.deb
```
I never use debian/XXXX unless I am building a deb and don't expect it to
supply a full set of packages for building, just to run packages.

The basic set of files you need are in debian/control.in

Will probably find that libprotobuf-dev, libprotoc-dev, protobuf-compiler and python-protobuf
will pull all the rest with them.

sirop, 2016-05-06:
@ArcEye 
I followed now your advice and built without `sudo mk-build-deps -ir` ,
but had to install these packages before:
```
 sudo apt-get install -y  libprotobuf-dev protobuf-compiler python-protobuf libprotoc-dev
 sudo apt-get install -y  libtool liburiparser-dev
 sudo apt-get install -y  libssl-dev
 sudo apt-get install -y  libavahi-client-dev
 sudo apt-get install -y  libglib2.0-dev
 sudo apt-get install -y  libudev-dev
 sudo apt-get install -y  libgtk2.0-dev tcl8.6-dev tk8.6-dev
 sudo apt-get install -y  libxaw7-dev
 sudo apt-get install -y  libreadline-dev
 sudo apt-get install -y  tcl8.6-dev tk8.6-dev
 sudo apt-get install -y  tcl8.6 tk8.6
 sudo apt-get install -y  python-tk
 sudo apt-get install -y  libboost-python-dev
 sudo apt-get install -y  libglu1-mesa-dev
```

Well, then I'll do it this way in future.

ArcEye, 2016-05-06:
> libreadline-gplv2-dev got me into dependency hell. It should have been libreadline6-dev from the very beginning.

I think this is largely a debian package problem.

They have both https://packages.debian.org/jessie/libreadline-gplv2-dev
and
https://packages.debian.org/jessie/libreadline6-dev
in the same repo
The meta-package *libreadline-dev* requires *libreadline6-dev* and conflicts with *libreadline-gplv2-dev*
so it is possible to get in a right state, depending upon what was installed first.

I think the `libreadline-gplv2-dev` requirement carried over from Wheezy, but could now just be replaced with the `libreadline-dev` meta-package to ensure the correct versions get pulled.

I was going to edit my original list of Jessie building and development packages in the docs, but that whole list appears to have been removed anyway.

sirop, 2016-05-06:
@ArcEye 
>I was going to edit my original list of Jessie building and development packages in the docs, but that whole list appears to have been removed anyway.  

Do you mean this list:
https://github.com/luminize/machinekit-docs/blob/master/machinekit-documentation/developing/machinekit-developing.asciidoc

ArcEye, 2016-05-06:
> Do you mean this list:
https://github.com/luminize/machinekit-docs/blob/master/machinekit-documentation/developing/machinekit-developing.asciidoc

I did, but it is only in @luminize 's repo.  It is not in the main one, not even in my fork, because I updated it recently
https://github.com/machinekit/machinekit-docs/blob/master/machinekit-documentation/developing/machinekit-developing.asciidoc

sirop, 2016-05-15:
I looked into it again and used
```
vagrant@debian-jessie:~/machinekit$ sudo apt-get build-dep machinekit
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'libncurses5-dev' instead of 'libncurses-dev'
The following packages have unmet dependencies:
 libboost-thread-dev : Depends: libboost-thread1.55-dev but it is not going to be installed
E: Build-dependencies for machinekit could not be satisfied.
```
adding `deb-src http://deb.machinekit.io/debian jessie main` to ` /etc/apt/sources.list.d/machinekit.list` before.

As you see there is a dependency problem: `libboost-thread1.55-dev` depends on `libstdc++`,
which is already installed as a  _sid_ and a _jessie_  version:
```
vagrant@debian-jessie:~/machinekit$  apt-show-versions | grep libstdc++
libstdc++-4.9-dev:amd64/jessie 4.9.2-10 uptodate
libstdc++6:amd64/sid 6.1.1-3 uptodate
```

Postponing the installation of `libczmq-dev/sid` right until `./configure` helps for a while,
but then `make` yields an error compiling `protobuf`.  

So if the manual installation of supporting packages as in https://github.com/machinekit/machinekit-docs/issues/71#issuecomment-217413049 is not elegant enough, what is to do?
Build  `libczmq-dev´ for jessie?

sirop, 2016-05-15:
Anyway, just built `libczmq-dev (3.0.2-7)` on jessie with `./src/czmq_selftest` passing all tests.

sirop, 2016-05-16:
Built machinekit with libczmq-dev (3.0.2-7) on jessie/amd64 .

Had to
```
 sudo apt-get install --only-upgrade libwebsockets-dev
```
in between, but the good news is that `sudo mk-build-deps -ir` now really pulls in all _Build-Depends_ packages.  

So what is next to do? Ask the maintainers of https://packages.debian.org/en/sid/libczmq-dev
to include libczmq-dev also into jessie?

sirop, 2016-05-19:
Luca Boccassi (one of the Debian libczmq maintainers) promised to include libczmq-dev (3.0.2-7)
into jessie-backports.  

Until then the following should be a workaround to avoid overlapping _libstdc++_ dependency between jessie and sid:
```
wget http://http.debian.net/debian/pool/main/c/czmq/czmq_3.0.2.orig.tar.gz
tar -xfv czmq_3.0.2.orig.tar.gz
cd czmq-3.0.2
./configure
make
make check
sudo make install
```

You may need 
```
sudo apt-get install -y build-essential libtool pkg-config autotools-dev \ 
autoconf automake cmake uuid-dev libpcre3-dev
```
as well as
```
sudo apt-get install -y libzmq3 libzmq3-dev
```
if not already installed.