machinekit/machinekit-docs #118: formatting and style of the website could be better

mhaberler, 2016-05-30:
this is an anchor issue about the web mechanics perspective:

- which style/fonts/layout/css etc should we use?
- which source highlighting style is best
- do we need an epub format of the website?

we need a bit of a style guide for writing documents and man pages!

luminize, 2016-08-08:
@mhaberler I think you can tick off "source highlighting". Regarding style guide, how about something like this for the manpages https://github.com/luminize/machinekit-docs/commit/0c6199ee1116e3b091346497438b5929b1524c7e (to get some feeling on usability / readability) or @ArcEye does this completely mess up all the work that you already did with instcomp and mank ?

* using tables gets a (IMO) better structured overview of the pins, type, direction and description
* syntax makes usage clear

ArcEye, 2016-08-09:
It won't mess up instcomp and comp, because I won't be changing it !
All of man3 and man9 need to display properly before I even think of changing something that produces a fully readable page currently.
Tables are catered for specifically, but there are only 2 or 3 components that use them.

Mank just displays a text rendered html page converted from the asciidoc.
I seem to remember having to change the existing tables so that they displayed correctly in a browser AND mank, so you will need to check that.

luminize, 2016-08-11:
It wasn't necessarily trying to change things, more as a thought on how to better display information. See https://snag.gy/06E5va.jpg and https://snag.gy/w2TkP4.jpg should give you an idea without having to render said branch.

ArcEye, 2016-08-12:
Those look good, but you would need to test the html in `elinks` (or open the asciidoc with mank) to see how they are rendered.