machinekit/machinekit-docs #121: QtQuickVCP GUI is nowhere mentioned

luminize, 2016-05-30:
the use and setup of QtQuickVCP is not mentioned in the docs. Cetus and Machineface for example could be added to (future separate) section of GUI's. But general instructions on making a custom UI, setting up the Machinekit SDK or vagrant would lower barrier of use

mhaberler, 2016-05-31:
@strahlex - we need to start discussing how to bring QtQuickVCP docs here, and the whole project into a support able shape.

Can you make some proposal?

strahlex, 2016-06-20:
Aye, some generic way to document Machinekit UIs needs to be found. Is there a way to fetch the docs from the project repository? From a maintenance perspective it makes sense to keep documentation and project close. The best approach for documenting smaller projects so I have seen so far is the monolithic README. Would it be possible to use this README e.g. from QtQuickVcp or Machineface as documentation inside the Machinekit docs?

ArcEye, 2016-06-21:
> Would it be possible to use this README e.g. from QtQuickVcp or Machineface as documentation inside the Machinekit docs?

Unfortunately markdown translates badly to asciidoc, as we found during our conversion of the site docs.

Github does not want you to display stuff elsewhere, so does not care, .md works OK for them :smile: 

You can use the document as a basis, but there is a fair bit of manual alteration required to get the format which displays best on the site through Jekyll.

Here is one I prepared earlier to get you started. #175

ArcEye, 2016-06-21:
There is a slight glitch, whereby the link shows sometimes in the right hand `Developing` menu bar
and then disappears when a different page is displaying.

I am playing with this now locally, but I think it is the `incremental build cache effect` we struggled with all through the site rebuild and will just require a forced build of `_site` to get it right again.

Yeah, that was it, I will go into Jenkins and delete the workspaces to force a complete rebuild and all will be rosy:)

ArcEye, 2016-06-22:
@strahlex the document is up on the site now
http://www.machinekit.io/docs/developing/qtquickvcp

You will need to check the links and amend them as required

Some of the ones that are redirected to 'thecooltool' repo are coming up as `404`
https://github.com/strahlex/TCT3D/blob/revision2/UNIPRINT-3D.ini#L86
https://github.com/strahlex/TCT3D/blob/revision2/UNIPRINT-3D.hal#L610
as is
http://static.machinekit.io/html/qtquickvcp/index.html

Not necessarily an exhaustive list, but they were the ones that stood out