machinekit/machinekit-docs #143: The goal of the project is not clear

luminize, 2016-05-31:
People think that machinekit is a beaglebone or a 3D printer project.
The landing page therefore is not clear enough.

http://blog.machinekit.io/2015/11/summary-of-open-discussion-at-nov-2015.html?m=1

Gives a good summary.
The landing page should better explain the project

- reusable stack to build applications on top
- multi platform
- HAL
- realtime
- remote UI

ArcEye, 2016-06-04:
Means moving away from the 'Moves Controls Things' so will require a consensus.

There can only be one message and one landing page, so it needs to be the first thing seen.

I share your frustration regards BBB and 3d printing.  
I have no interest in either but if you tell people they would find an x86 computer far better and probably cheaper plus easier to use, for what they want to do, they think you are a dangerous heretic.

luminize, 2016-06-04:
> On 04 Jun 2016, at 14:12, ArcEye wrote:
> Means moving away from the 'Moves Controls Things' so will require a consensus.
> 
> There can only be one message and one landing page, so it needs to be the first thing seen.
> 
I agree, I think "moves, controls, things" can still be a title thing, but I would make a very clear, unambiguous message. Explaining the "things" now is too general IMO. It's the 30 second elevator pitch which should be on the landing page.

cdsteinkuehler, 2016-06-04:
On 6/4/2016 10:29 AM, Bas de Bruijn wrote:
>  > On 04 Jun 2016, at 14:12, ArcEye wrote:
>  >
>  > There can only be one message and one landing page, so it needs to be the 
> first thing seen.
>  >
> I agree, I think "moves, controls, things" can still be a title thing, but I 
> would make a very clear, unambiguous message. Explaining the "things" now is too 
> general IMO. It's the 30 second elevator pitch which should be on the landing page.

IMHO, the project goal is stand-alone HAL, a foundation that takes
care of the difficult and complicated bits about doing hard-real-time
and allows other developers to focus on their project (traditional
CNC, 3D printing, robot control, autonomous driving, whatever), while
being (mostly) platform agnostic thanks to Linux.

The legacy CNC stack and 3D printer support should not be abandoned,
but I don't think it's the core focus of the project.

-- 
Charles Steinkuehler
charles@steinkuehler.net

strahlex, 2016-06-05:
What about Machinetalk? IoT and M2M support is what gets you management attention these days. I think HAL Remote is an important selling point.

strahlex, 2016-06-05:
When it comes to convince people from the industry to use Machinekit, there are some other aspects that need to be explained. Especially important is to point out what Machinekit does different than typical RTOS systems for micro-controllers. There needs to be a good reason to use a full-blown Linux system over a special purpose OS which runs on a < 1€ chip. Key points are:

- Clear standardized platform
 - Component-based system
 - Real-time
 - Time triggered (are we there yet with event triggered "threads"?)
- Run-time reconfigurable
- Use the full potential of Linux

Applications are also important, for the classical control systems application Machinekit is probably to "big" (again, it does not fit on a 10c uC). Therefore, we should focus on the use cases where Machinekit makes sense, which are all applications that require an Embedded Linux platform anyway:
- Robotics -> as motion controller
- CNC/3D printing -> as motion controller
- Prototyping -> control systems, ...
- As high-level PLC, with IoT/M2M capability -> UPC-UA anyone?
- Learning, studying, experimenting

cdsteinkuehler, 2016-06-05:
On 6/5/2016 12:53 PM, Alexander Rössler wrote:
> What about Machinetalk? IoT and M2M support is what gets you management 
> attention these days. I think HAL Remote is an important selling point.

I agree about HAL Remote, it's huge and fits in exactly with what I
think is the core Machinekit mission (re-usable HAL layer to make
real-time stuff easy).

Machinetalk is a bit different.  I personally love the idea, and am
excited by the option to have a remote UI that is not based on
forwarding X11 traffic or using VNC, but at the same time it's a UI
layer on top of a (legacy and fairly convoluted) CNC stack.  I think
Machinetalk is a key feature (especially for folks on display or GPU
constrained platforms like the BeagleBone and the new SoC+FPGA boards
that typically don't have anything but a serial console), but I'm not
sure it's part of the 30 second Machinekit "elevator pitch".

I guess it depends on how much of the CNC stack we're officially
adopting, and whether or not the CNC stuff remains part of Machinekit
proper, or becomes something like machinekit-cnc, which runs on top of
machinekit-hal.

Thoughts?

-- 
Charles Steinkuehler
charles@steinkuehler.net

strahlex, 2016-06-13:
@cdsteinkuehler HAL Remote is part of Machinetalk. I think the misunderstanding is that Machinetalk is just for user interfaces whereas it is actually a middleware for distributed systems. Not sure if we should explicitly mention Machinetalk (because the name is easy to confuse with Machinekit) or just IoT/M2M support.

sirop, 2016-06-17:
@strahlex 
>When it comes to convince people from the industry to use Machinekit, there are some other aspects that need to be explained.   

One thing that can frighten away the industry is the licence issue.
MK consists mostly of GPL.  
So can any HAL component have a different licence, not GPL?
At least LGPL somehow?