machinekit/machinekit-docs #144: Website text boxes have poor contrast

cdsteinkuehler, 2016-05-31:
I am having problems reading the default textboxes rendered on the website.  The black on dark-gray and turquoise on dark gray have very little contrast.  Is this what it looks like for everyone else?

[image]

mhaberler, 2016-05-31:
same here

we need to reconsider site style, and that is just  one aspect

font - looks flashy, readabiliy: poor IMO
screen real estate efficiency - poor, narrow center column, lots of wasted space
shading, quoted text background color - not so great

luminize, 2016-05-31:
@cdsteinkuehler @mhaberler Yes, I agree.

The green/yellow of the logo is not my personal taste, but I can live with that.
What is less attractive is the yellow background of code blocks. Plus literal `code` in the text is another shade of yellow. 

Like looking into a bowl of custard ;)

My take is "less is more" good reading is more important than rounded corners and colors.

We can search a bit on Jekyll templates and see what's out there to form an opinion.

ArcEye, 2016-06-03:
PR #157 goes a long way to resolve

Highlights now custard-cream not custard-vomit
[image]

Tables now high contrast background
[image]

[NOTE] a peaceful graduated blue
[image]

[WARNING] a bit more strident but not glaring
[image]

ArcEye, 2016-06-03:
Wrong button

PR #157 (2016-06-03): Modify style sheet to achieve proper contrast between text and bgrnd
Existing style rendered backgrounds in muddy grey with black test,
which made for difficult reading.
Tables and Warnings were particularly bad

All graphic text backgrounds changed to a high contrast pale violet.

[NOTE] colours changed to a more restful pale blue

[WARNING] upgraded to the same type of banner as [NOTE] with muted red
background

To see all above plus [source] highlighting, see
preview.machinekit.io/docs/config/ini_config
which contains the lot.

Goes a long way to fix #144
commit 2d528273 (2016-06-03): Modify style sheet to achieve proper contrast between text and background.

Existing style rendered backgrounds in muddy grey with black test,
which made for difficult reading.
Tables and Warnings were particularly bad

All graphic text backgrounds changed to a high contrast pale violet.

[NOTE] colours changed to a more restful pale blue

[WARNING] upgraded to the same type of banner as [NOTE] with muted red
background

To see all above plus [source] highlighting, see
preview.machinekit.io/docs/config/ini_config
which contains the lot.

Goes a long way to fix #144
closed 2016-06-03 by ArcEye