machinekit/machinekit-docs #168: Elaborate on Beaglebone Black cape resource conflicts

HakanBastedt, 2016-06-18:
Just got my first ever machinekit up and running. One part I didn't find in the documentation was how to resolve resource confllcts between capes.
It manifested itself as 
```
Loading cape-universal overlay
bash: line 0: echo: write error: File exists
```
It isn't all that easy to understand what the problem is. dmesg gives the right clues though.

From what I understand this is a well known issue (now when I have googled it).

The solution is to load/unload HDMI/HDMIN cape.
I think a description of this topic would be a great fourth point under the BeagelBone Black / jessie installation.
Ref: https://groups.google.com/forum/#!topic/machinekit/AjBFY6aMSeQ

luminize, 2016-06-18:
How about a general problem solving section FAQ on peripherals? This isn’t a Machinekit specific problem (mind: this is not a BeagleBone project) but there could also issues that arise on PC or other devices which are not Machinekit specific, but nevertheless will speed up things if you’re aware of them.

ArcEye, 2016-06-18:
Please cut and paste the thread you refer to, down to a text, which describes the problem and gives a solution.
There is a lot of verbiage in there that is not related.

Then go here http://www.machinekit.io/docs/sandbox/index.html and follow the instructions for creating a sandbox topic

Once the article is posted we will look at where best to put it.

A platform related FAQ as @luminize suggested, might be the answer

luminize, 2018-10-09:
Nothing done in 2 years. Closing due to lack of interest. This seems to be no real problem.

closed 2018-10-09 by luminize