machinekit/machinekit-docs #177: FDM/FFF (3D printing) documentation

strahlex, 2016-06-24:
There should be a section about how to use Machinekit for 3D printing. The FDM standard should also move into the docs: https://github.com/strahlex/Machinekit-FDM-Standard

luminize, 2016-06-24:
That would only make sense imo if the configurations are "stock" machinekit.

the-snowwhite, 2016-06-24:
I Agree with you both. For mere 3d printing is not just a "toy" I have seen
3d Glass printing and 3d Bridge building with robot welders.

But this is not My Call as I was not part of the deciding around the
externalisation of the qt stuff.

luminize, 2016-06-24:
> On 24 Jun 2016, at 19:13, Michael Brown <notifications@github.com> wrote:
> 
> But this is not My Call as I was not part of the deciding around the
> externalisation of the qt stuff.

I'm sorry, I don't see the connection. Qt has nothing to do with configurations or HAL

cdsteinkuehler, 2016-06-24:
On 6/24/2016 10:12 AM, Bas de Bruijn wrote:
> That would only make sense imo if the configurations are "stock" machinekit.

I think Alexander's standards for FDM setups and his example
configurations should migrate into the machinekit repo and become part
of the standard release.  IMHO, the existing FDM examples should also
migrate to using the FDM standards and possibly the python HAL setup.

-- 
Charles Steinkuehler
charles@steinkuehler.net

the-snowwhite, 2016-07-15:
What i remember the issue was about not adding more gui  stuff at that time, because of the build system being very strained.

I tried to add some qvcp configs

strahlex, 2016-07-16:
To clarify: The config itself is not UI specific, I See no problem in adding additional configs to the Machinekit repo.

cdsteinkuehler, 2016-07-16:
I agree with adding 3DP content to the docs as well as adding some stock configurations that use the FDM-Standard and/or migrating the existing 3DP configs to use the FDM-Standard.