machinekit/machinekit-docs #178: There are still issues in github.machinekit.io

luminize, 2016-06-24:
Since all documentation, including the source for the site now reside in machinekit-docs, and machinekit.github.io is only used for displaying the site, should we move/close issues on machinekit.github.io to machinekit-docs, and disable the possibility of making issues on machinekit.github.io repo?

https://github.com/machinekit/machinekit.github.io/issues/60
https://github.com/machinekit/machinekit.github.io/issues/61

mhaberler, 2016-06-25:
yes we should remove the tracker from this repo, and refer to mk/mk-docs

luminize, 2016-08-08:
relevant issues have been brought into this repo. I have disables the possibility to file issues at machinekit/machinekit.github.io

closed 2016-08-08 by luminize