machinekit/machinekit-docs #244: packages install in different sections in development setup instructions

luminize, 2016-11-03:
I noticed when prepping a new BBB image for development that instructions to install packages are here: 
http://www.machinekit.io/docs/developing/machinekit-developing/#install-development-packages
specifically:
```
sudo apt-get install libczmq-dev python-zmq libjansson-dev pkg-config \
  libwebsockets-dev libxenomai-dev python-pyftpdlib cython bwidget lsb-release
```

now when getting to the section:
http://www.machinekit.io/docs/developing/machinekit-developing/#get-source-and-build
in the step `sudo mk-build-deps -ir` it pulls in a lot of packages. 

2 questions:
1. Can't the packages above be pulled in in the same way via `sudo mk-build-deps -ir`?
2. Am I right that the package `machinekit-dev` is missing in above instructions ( I get the impression that this is needed for the `sudo mk-build-deps -ir` step?

zultron, 2016-11-03:
@luminize,

1.  Yes, `sudo mk-build-deps -ir` (after `./debian/configure`) will install all strict build deps, sufficient for RIP and package builds.  There are also a few packages that are run-time deps but not build deps that must be installed separately.

2.  The `machinekit-dev` package is a package build result, not a package build dependency, so it's not needed by `mk-build-deps`.

luminize, 2016-11-03:
@zultron 

re. 1:
so adding said packages to ?some file? will pull these in with mk-build-deps ?

re. 2:
I first uninstalled machinekit packages from an SD image with `sudo apt-get remove machinekit --purge` but I couldn't run `sudo mk-build-deps -ir` until I first installed the machinekit-dev package. Does that make sense?

zultron, 2016-11-04:
@luminize,

For 1, `mk-build-deps` only installs build deps, not run-time deps.  I haven't explored how to install run-time deps automatically, beyond installing packages, of course.  It does seem like the docs ought to describe just one method or the other.  Is that what you're driving at with this issue?

For 2, I think you should be able to run `mk-build-deps` without the `machinekit-dev` package installed, following the instructions in the second link you pasted above.  Can you post your error message?

luminize, 2016-11-04:
for 1:
no, not particularly, I wasn't aware on the difference on the build deps and run-time deps. So that seemed like it would be easier to pull in the run-time deps at the same time as the build deps.
for 2:
I think I must have done something wrong while working from 2 ssh shells, doing different things. All works as per instructions on the docs

thanks

closed 2016-11-04 by luminize