machinekit/machinekit-docs #253: List project maintainers on website

machinekoder, 2017-03-14:
Currently, there is no place where all project maintainers are publically listed. I suggest adding a new section on the Machinekit.io website containing the project maintainers and where to find people to contact for commercial services.

Currently, the only place to find a list of project maintainers is here:
https://github.com/orgs/machinekit/people
However, even on this page, many people are marked private.

ArcEye, 2017-03-14:
Originally what you said included
> ...and where to find people to contact for commercial services.

If you want to advertise yourself for hire that is one thing.

I deliberately keep my internet profile low and only take contacts through the official channels (github and google groups).
I have no interest in marking up extra contact details so that people who want to try and bypass the system can try and contact me directly.

the-snowwhite, 2017-04-28:
Mnnnn .