machinekit/machinekit-docs #254: Add Commercial Support section to website

machinekoder, 2017-03-14:
The Gtkmm website has a page called [commercial support](http://www.gtkmm.org/en/commercial_support.html) where people can list themselves if they are interested in providing commercial services evolving around the project.

Since people are already using Machinekit in commercial products, I suggest adding a similar page to the Machinekit website.

ArcEye, 2017-03-14:
Fine by me, you can reuse the space in the headings taken by the 'Sandbox'.

No-one has used that, in the same way that virtually no-one has used the 'Edit This Page' facility.

ArcEye, 2017-03-18:
https://github.com/machinekit/machinekit-docs/pull/255

Fill your boots!

closed 2017-03-18 by ArcEye

machinekoder, 2017-03-18:
Aye!

ArcEye, 2017-03-19:
There is a problem with the website builds.
The preview site works fine, but the last 10 builds of the website proper have failed.
This happened before, need to try and remember what was needed to kick start it again.
Have a look later.