machinekit/machinekit-docs #303: Website source has 15k+ blank lines

darenschwenke, 2019-02-23:
View source: www.machinekit.io/docs/man/man9/kins/
It's nearly 30k lines long.  I didn't count, but I'm guessing at least 50% of that is whitespace/newlines.
I think perhaps consecutive blank lines need to be stripped on save if not inside a strict formatting section.

ArcEye, 2019-02-23:
Afraid I really don't see the problem.
The spacing makes it clear and easy to read.

As for being 30,000 lines long :confused: 

kins.asciidoc is 113 lines long and only 3343 bytes.

cdsteinkuehler, 2019-02-23:
On 2/23/2019 5:36 AM, ArcEye wrote:
> Afraid I really don't see the problem.
> The spacing makes it clear and easy to read.
> 
> As for being 30,000 lines long :confused: 
> 
> kins.asciidoc is 113 lines long and only 3343 bytes.

I see the problem.  The html rendered by the page is 29652 lines long,
most of which is a bunch of blank lines in the <ul> lists which make
up the menu on the right-hand side of the page.

The HTML for the kins itself looks sensible and is less than 200 lines.

...so the issue isn't really with kins, it's with the web navigation menu.

-- 
Charles Steinkuehler
charles@steinkuehler.net

ArcEye, 2019-02-23:
OK, I see.

The html page is generated, sometimes several times a day, from the asciidoc and several css style sheets plus the added html content to create the overall page.

There is no human input outside of the asciidoc contents.
So long as the page looks how it should, the size of it is irrelevant to us and not something that can be edited.

darenschwenke, 2019-03-26:
Ok.  Just made it _quite_ a bit harder when I went to go find HTML targets I could use for pointing people directly to a certain tagged section.  :) Thought you should know.

closed 2019-03-28 by ArcEye