machinekit/machinekit-docs #321: Strugling to find any link to new EMCApplication project build docs

Holotronic, 2020-10-12:
[from PR](https://github.com/machinekit/machinekit-docs/pull/320/files)
There seems to be missing a link in the Docs to any file in docs/developing refereed to by [this file:](https://github.com/machinekit/machinekit-docs/blob/master/docs/developing/develop-index.asciidoc)

Holotronic, 2020-10-12:
I think the link on [this page:](http://www.machinekit.io/docs/index-developer/) at the bottom named:
Setting up development environment --> http://www.machinekit.io/docs/setting-up/developing-setting-up

Needs to be replaced by:
Setting up development environment --> http://www.machinekit.io/docs/developing/developing/

And then http://www.machinekit.io/docs/setting-up/developing-setting-up/ page removed
any links referring to http://www.machinekit.io/docs/setting-up/developing-setting-up/ page should instead
refer to -->  http://www.machinekit.io/docs/developing/developing/

cerna, 2020-10-13:
@the-snowwhite,
thank you, you are right.

Problem is, the whole Machinekit website (documentation) is pretty bad. (I think this is not the only unfindable page there.) Another problem is that so far the documentation recommends installing the `deb.machinekit.io` repository, but Machinekit is now using the Cloudsmith repository (because nobody had at the time keys to the old one, it is hosted on personal server of _gone fishing_ Machinekit member and granting access is now much simpler). I was thinking of keeping it that way as the situation is now pretty volatile and the old _monorepo_ Machinekit package was stable, but maybe with addition of EMCApplication (CNC stack) to Machinekit organization, it is time to remove the old repository and propagate the new one.

What do you think?

the-snowwhite, 2020-10-14:
@cerna 
Well I'd love to slash the old `deb.machinekit.io` repository[ execpt for this issue](https://github.com/machinekit/machinekit-hal/issues/324)

the-snowwhite, 2020-10-18:
@cerna 
also the socfpga-rbf package needed to run anything hostmot2 related on the armhf soc's is completely missing from cloudsmith
Can the socfpga-rbf and socfpga-bit be copied over to the mk-hal cloudsmith repo ?

cerna, 2020-10-18:
@the-snowwhite,
I have been worried about the FPGA package buildings from the moment the Jenkins server went down. Because it is resource hog. (And I - so far - have no experience with building and deploying FPGA solutions, _but what isn't can be..._) **So far I wasn't able to find public CI/CD service for Open-Source projects which would allow building it.**

That `socfpga-rbf` package is a dependency? So it goes into [Machinekit/Machinekit](https://cloudsmith.io/~machinekit/repos/machinekit/packages/)? Or should it go into the Machinekit/mksocfpga repository. (I have been following in Cloudsmith the same pattern as in Github Machinekit organization.)

_BTW, I will add you to the Cloudsmith organization. Do you want to use the same e-mail you are using when commiting?_

the-snowwhite, 2020-10-18:
socfpga-rbf/bit --> Machinekit/mksocfpga (on cloudsmith)
as e-mail regards my commit e-mail should be in the line of: pro...ic.xx