machinekit/machinekit-docs #325: Machinekit Docs not updating The full error is:

the-snowwhite, 2021-03-24:
The full error is:
Run str="machinekit/machinekit-docs"
Error: Unable to process command '::set-env name=PARSED_USERNAME::machinekit' successfully.
Error: The `set-env` command is disabled. Please upgrade to using Environment Files or opt into unsecure command execution by setting the `ACTIONS_ALLOW_UNSECURE_COMMANDS` environment variable to `true`. For more information see: https://github.blog/changelog/2020-10-01-github-actions-deprecating-set-env-and-add-path-commands/
Error: Unable to process command '::set-env name=PARSED_LOCAL_REPOSITORY::machinekit-docs' successfully.
Error: The `set-env` command is disabled. Please upgrade to using Environment Files or opt into unsecure command execution by setting the `ACTIONS_ALLOW_UNSECURE_COMMANDS` environment variable to `true`. For more information see: https://github.blog/changelog/2020-10-01-github-actions-deprecating-set-env-and-add-path-commands/

_Originally posted by @the-snowwhite in https://github.com/machinekit/machinekit-docs/issues/323#issuecomment-805994311_

cerna, 2021-03-26:
Thank you, @the-snowwhite, it looks like the Github Actions workflow has rotten a bit in the meantime. I will try to repair it over the weekend.

cerna, 2021-04-04:
@the-snowwhite,
the solution for this problem comes in two packages:

- Pull request machinekit/machinekit.github.io#65
- Pull request machinekit/machinekit-docs#326

The pull request machinekit/machinekit.github.io#65 should be merged first.

cerna, 2021-04-04:
Seems to be working now.

closed 2021-04-04 by cerna